TheSpaghettiDetective/OctoPrint-Obico #168: [Feature] Better pause awareness and action from inside Octoprint

toddmedema, 2022-05-29:
Problem: The pause experience from inside Octoprint is confusing and lacks context.

**Describe the solution you'd like**
When a pause is triggered in Octoprint:
1. There should be a **clear indicator of what triggered the pause**; perhaps as a toast message that must be dismissed, and links to Spaghetti Detective UI for more info
2. **Unpausing from Octoprint should be a valid action**. Right now doing so causes incorrect printing and immediate re-pausing. Ideally, clicking unpause from Octoprint would behave similarly to unpausing in Spaghetti Detective, where it prompts you to ask if you want to disable detective for the rest of the print. 

**Describe alternatives you've considered**
1. Rather than (or in addition to) a toast, you could also add a new status "Paused by Spaghetti Detective" that would be displayed in the status widget, so that you at least know the cause
2. If it's not possible to make the resume button functional in Octoprint, I would suggest disabling it and linking to Spaghetti Detective, rather than allowing users to accidentally do something that doesn't work

**Additional context**
We're in a shared workshop where one person configured the 3D printing setup, but plenty of other people use it. Thus the lack of context around "why does this print keep pausing?" and "what do I do about it?"

kennethjiang, 2022-05-31:
Sorry for taking so long to get back to you on this.

For your 1st point: by "what triggered the pause", do you mean indicating Obico failure detection vs all other possible reasons why a print may be paused? If so, we are actually about to implement an event list that shows a chronological list of all events, including a failure is detected and printer paused. Will that serve the purpose?

For your 2nd point: it sounds like a bug. Can you open a separate issue with a description on how to reproduce it? By design, resuming a print in Obico should behave exactly the same as resuming it in octoprint. Any behavior deviating from that is a bug.

toddmedema, 2022-05-31:
No worries! I didn't expect anyone to respond over a holiday weekend :)

1. That sounds awesome! Will that be shown in Octoprint, or just the Spaghetti UI? If it's in Octoprint, that should be perfect :) (bonus points if it links to your Spaghetti UI URL)
2. Not a bug - they both behave the same today. Aka if you hit "resume" from either UI, it'll try printing again for a bit, then re-pause when it detects the same spaghetti. Thinking more about it, I think that as long as the first point is solved well, this behavior is fine since you'll know to go to Spaghetti and can investigate there
3. Context: We have a continuous belt printer tucked away on another floor that's generally been pretty reliable, so if Spaghetti falsely triggers on a print and the first few layers have stuck successfully, our first reflex is generally to disable for the rest of the print to prevent further pauses. This may by different than how other / most users of Spaghetti use it :)

kennethjiang, 2022-06-01:
It'll be only shown in the Obico app. The reason why we don't want to show the event list in OctoPrint is this list is only a small set of events OctoPrint generates, mixed in a few events generated by the Obico app. We are afraid that showing this event list may mislead many users to think this is a list of OctoPrint events.