TheSpaghettiDetective/OctoPrint-Obico #170: [BUG] custom server URL validation fails on hostname URLs

smartin015, 2022-06-23:
**Describe the bug**

**To Reproduce**

1. Go to octoprint instance, with fresh installation of Obico plugin
2. Click "Custom server" radio button, type e.g. `http://hostname:3334/`, click Continue

Rather than continuing to the next screen, an `Invalid server address` error appears and the wizard cannot continue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser: Chrome
 - Version: 102.0.5005.75 

**Additional context**

This is a regression from previous TSD behavior where I was successfully able to connect to my local server with `http://hostname:3334`.

The [regex you're using for URL validation](https://github.com/TheSpaghettiDetective/OctoPrint-Obico/blob/10ca7760c1fa2946d50d2190b1824a1e69706504/octoprint_obico/static/js/ObicoWizard.js#L47) requires that there's a [TLD](https://en.wikipedia.org/wiki/Top-level_domain) after the hostname. While this is *technically* correct, the TLD can be left implicit when resolving hostnames on a local network. A host named `host1` for instance, can be resolved with `http://host1:3334`, although it may sometimes be implicitly routed as e.g. `http://host1.local:3334` (I'm a bit fuzzy on this - appending `.local` appears to work here, but IIRC windows prefers `.lan`... it gets confusing).

I recommend changing your regex to:

```
^https?:\/\/(www\.)?[-a-zA-Z0-9@:%._\+~#=]{1,256}\b([-a-zA-Z0-9()@:%_\+.~#?&//=]*)$
```

Which is more permissive and lacks the TLD constraint, bringing the process back to parity with pre-Obico.

kennethjiang, 2022-06-29:
Good point. Feel free to send a PR for it and I'll merge it.

Error-418, 2025-02-24:
This is still an open issue as of 2/23/2025. I know because I had to find this issue to fix the problem on my self-host, as the instructions to include **http://** aren't in the setup instructions. I tore my hair out for an hour trying to figure out the issue, I am now bald.

Please either fix the instructions or fix the program to properly sanitize the server pathway, as a QoL issue.