TheSpaghettiDetective/OctoPrint-Obico #174: [BUG] FFMPEG bug causes Obico not to respect RFC2396 for stream URLs causing compatible mode to run at 0.1 fps

puterboy, 2022-06-24:
In Octoprint settings, under "Webcam & Timelapse", if you set stream url to:
`http://myrpi:8080/?action=stream` which is a valid RFC2396 URL, then ffmpeg will fail to recognize the URL and returns:
`http://myrpi:8080?action=stream: Server returned 400 Bad Request`

This is actually an ffmpeg bug
See the bug report: https://trac.ffmpeg.org/ticket/8466

Obico doesn't detect this bug and merrily launches ffmpeg with the faulty URL.
This causes ffmpeg fails (silently) and then Obico defaults to 0.1fps snapshots.

Suggested fix is either:

1. Warn users if 'malformed' URL (and document to prevent users like me from wasting hours of time and having to debug the code to figure out what is going on)
2. Or add a manual fix to stream_url in webcam_stream.py to insert a slash if missing before '?'

Of course, until ffmpeg or Obico code is adjusted, users can avoid the problem by making sure there is a "/" before the '?'.
i.e., change `http://myrpi:8080?action=stream to http://myrpi:8080/?action=stream`

kennethjiang, 2022-06-25:
Good catch. Feel free to send a PR for it.

puterboy, 2022-06-27:
I'm not really a Python programmer, but I think this should work (feel free to correct my crude "style")

```
--- webcam_stream.py    2022-06-26 22:23:20.631896668 -0400
+++ webcam_stream.py.new   2022-06-26 22:30:05.722292999 -0400
@@ -192,6 +192,9 @@
         jpg = wait_for_webcamd(webcam_settings)
         (_, img_w, img_h) = get_image_info(jpg)
         stream_url = webcam_full_url(webcam_settings.get("stream", "/webcam/?action=stream"))
+       stream_url_parse = urlparse(stream_url)
+        if stream_url_parse.query and not stream_url_parse.path :
+            stream_url = stream_url_parse._replace(path='/').geturl()
```

puterboy, 2022-06-27:
Alternatively, you could add the patch to the definition of  webcam_full_url in

puterboy, 2022-12-21:
Any reason this patch (or similar) has not been included in latest updates?