TheSpaghettiDetective/OctoPrint-Obico #175: [Feature] Add button to Obico app/window to prevent/delay video streaming timeout

puterboy, 2022-06-24:
Currently, whenever I hide or close the window in which Obico is running and then return to the window, the video stream stalls while buffer as presumably the streaming was interrupted while the window wasn't active.

I understand why this feature may be important (even critical) when streaming remotely when bandwidth is limited/costly -- especially if using Obico cloud.

However, if you are running at home using your own local Obico-server, one should be able to override this feature and keep the streaming going all the time.

It's especially annoying when even a momentary alt-tab away from the Obico browser tab causes the stream to re-buffer -- which takes about 3-5 seconds.

Ideally, have options as follows in the Obico UI allowing for on-the-fly configuration of the following behaviors:
1. Current behavior - stop streaming when tab/app inactive
2. Stream forever
3. Stop streaming
4. Stream only when printing
5. Continue streaming for user-configurable X minutes after tab/app is inactive.

kennethjiang, 2022-06-25:
I can't reproduce it based what you described here Please list the steps to reproduce so that we can reproduce it.

puterboy, 2022-06-26:
I'm not doing anything special that I know of... but this is my current context. 
- Running Octopi + Obico on vanilla Raspbian on a RPI4 with 4GB
- Running Obico-Server on a Linux Dell Server
- I have an RPI camera working in Advanced mode (and restreaming jpg to Octoprint) (btw I am JK123 on the Discord board)

- Obico (whether run in a browser tab or as an Android app) experiences  a buffering delay (with white circle spinning) whenever the tab or app is activated/re-activated. The delay on the app is about 1 sec, in the browser about 3 seconds.
- The restreamed Octoprint stream shows no video delays when the 'control' tab is activated/reactivated, confirming that there is no problem with the streaming/restreaming on the RPI (as expected since the RPI is not aware of the status of the app/tab)
- This suggests that there is something either in the Obico app/browser code that pauses or causes the Obico server to pause when the tab/app is not active and then resumes (requiring restarting of the stream and hence buffering) when the tab/app is reactivated.
- I thought this was a "feature" in that this was done to save bandwidth when streaming via Obico cloud and hence logged this as a feature request (rather than bug) to allow for continuous streaming without pausing when the tab/app are inactive.

If this is not the intended behavior, happy to help troubleshoot.
In particular, is there any easy way to access the Obico server stream directly (without the Obico app) - can I view it in my browser directly? This would help separate my Obico server from the Obico client code running in the browser or app.

kennethjiang, 2022-06-27:
> * Obico (whether run in a browser tab or as an Android app) experiences  a buffering delay (with white circle spinning) whenever the tab or app is activated/re-activated. The delay on the app is about 1 sec, in the browser about 3

So the stream will resume after a brief period of showing the spinner, correct? If so, it is the expected behavior.

kennethjiang, 2022-06-29:
I'm not sure how many users will be willing to have the streaming going all the time just to cut a couple seconds of pause. Changing it is not hard. The relevant code is: https://github.com/TheSpaghettiDetective/obico-server/blob/release/frontend/src/components/StreamingBox.vue#L109 . Feel free to submit a PR that:

1. Makes a configurable feature to have always-on streaming.
2. Makes the configuration hidden from the main UI since it's not a frequent control for most users (unless you can prove otherwise), while discoverable enough.