TheSpaghettiDetective/OctoPrint-Obico #177: [BUG] Pause erroneously generates "Filament Change" event while true filament change generates repeat events every 50 seconds

puterboy, 2022-06-30:
Two related bugs with the same one line fix.

1. Pausing (whether from Octoprint or from Printer console) causes an erroneous email saying that "Your print job requires filament change."

2. An actual filament change event (e.g., M600), causes multiple repeat such emails every 50 seconds, rather than a single email.

The cause is due to the line in gcode_hooks.py:
```if "m600" in lineLower or "fsensor_update" in lineLower or "paused for user" in lineLower or "action:paused" in lineLower:```

In case #1, there are no more subsequent m600 gcodes or lineLowers that trigger this again.
In case #2, the printer (or at least my Prusa), sends the line ```echo:busy: paused for user``` every second, resulting in a new event being triggered every 50 loops or every 50 seconds as the variable `self.num_gcode_until_next_filament_change = 50` decrements.

So this explains the behavior.
Patch is simply to change the above line to:

```if "m600" in lineLower or "fsensor_update" in lineLower```

Patch is simply:
```
--- gcode_hooks.py       2022-06-29 13:37:23.000000000 -0400
+++ gcode_hooks.py.new  2022-06-30 13:01:04.456108044 -0400
@@ -33,7 +33,7 @@
             filament_change_event = True
         if line:
             lineLower = line.lower()
-            if "m600" in lineLower or "fsensor_update" in lineLower or "paused for user" in lineLower or "action:paused" in lineLower:
+            if "m600" in lineLower or "fsensor_update" in lineLower:
                 filament_change_event = True

         if filament_change_event:

puterboy, 2022-12-21:
Any reason why this patch (or similar) has not been included?
The duplicate email messages on pause are annoying :)

kennethjiang, 2022-12-22:
I have been tied up with other priorities.

Since you seem to be very close to having a fix. Can you send a PR, hopefully tested on your own printer?

puterboy, 2022-12-22:
I posted a simple patch above :)
Do you need it in another format?

kennethjiang, 2022-12-27:
Yes it'll be great if you can send it in as a PR. Also please do as much testing as you can on your end.

puterboy, 2022-12-27:
Just sent a PR.
My ultimate change was simpler and cleaner.
For reference the patch is: 
```
--- gcode_hooks.py.orig 2022-12-27 11:58:30.611436571 -0500
+++ gcode_hooks.py      2022-12-27 11:29:10.691600082 -0500
@@ -21,6 +21,7 @@
         return line

     def check_for_filament_change(self, gcode=None, line=None):
+        if gcode:
         self.num_gcode_until_next_filament_change -= 1
         if self.num_gcode_until_next_filament_change > 0:
             return
```

puterboy, 2023-01-20:
@Kevin - what is the status of this? Thanks

kennethjiang, 2023-01-21:
@puterboy Sorry for the delay. Have been really tied up with other tasks. But this is an important issue. So will probably get around to it next week.

TurnrDev, 2024-03-27:
I'm getting a filament change notification every 10 minutes for a print that was finished over 10 hours ago. Is there a fix?

kennethjiang, 2024-03-28:
> I'm getting a filament change notification every 10 minutes for a print that was finished over 10 hours ago. Is there a fix?

This issue was actually fixed (this is why you are not getting notifications every 50s). I'm closing this. Please open another issue.

closed 2024-03-28 by kennethjiang