TheSpaghettiDetective/OctoPrint-Obico #180: [BUG] Obico causes 409 CONFLICT error when loading files to print on Octoprint 1.8.1

deltreey, 2022-08-06:
**Describe the bug**
I updated Octoprint to 1.8.1 today, and immediately couldn't load any sliced files to print.  I saw the 409 error in my browser, and pulled the logs.  There I found this:

```
2022-08-06 04:11:29,472 - octoprint.plugins.obico - ERROR - 'NoneType' object has no attribute 'get_metadata'
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_obico/print_event.py", line 80, in get_file_metadata
    file_metadata = plugin._file_manager._storage_managers.get(origin).get_metadata(path)
AttributeError: 'NoneType' object has no attribute 'get_metadata'
```

Disabling the Obico plugin immediately resolved the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Obico 2.1.2 on Octoprint 1.8.1
2. Upload a file to print
3. Click on the Load or Load & Print button
4. Nothing happens.  You'll see a 409 CONFLICT network error in your browser's console if you keep it open.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Windows 11
 - Firefox
 - 103.0.1 (64 bit)

**Additional context**
I am happy to provide the full log if you like, but there wasn't any other useful information there.

kennethjiang, 2022-08-10:
Sorry. Not sure why I missed it. The 409 was returned by octoprint, not Obico, right? Can't tell it from the screenshot.

deltreey, 2022-08-10:
Correct.  This seems to be resolved in 1.8.2 though, so you can close the issue if you want.

kennethjiang, 2022-08-10:
Cool. Thank you for update. Sorry again for the delay!

closed 2022-08-10 by kennethjiang