TheSpaghettiDetective/OctoPrint-Obico #183: [BUG] Version 2.1.3 printer always offline self hosted Obico server

lettore, 2022-09-01:
**Describe the bug**
After upgrading to 2.1.3 I see the printer always offline in self hosted Obico server with https proxy, troubleshooting  gives 
Account status: Pro
Webcam streaming: Premium (advanced)
Connection to server: Okay
Connection to webcam: Okay
Test button gives Succeded

If I try to relink, Obico server can't detect Octoprint (but this is because is running in Docker not in host mode) but using the code it links properly.

Downgrading to 2.1.2 resolved the problem.

kennethjiang, 2022-09-02:
You need to [upgrade the Obico server as the server is not designed to be forward compatible](https://discord.com/channels/614543405724205137/661368008781987841/1011353104727420938). Sorry! :(

I'll leave this ticket open as I suspect there will be other self-hosted Obico servers running into the same problem.

lettore, 2022-09-02:
Well this is huge problem, there's a troubleshooting page that looks at server connection and test it saying is all OK. It will be so hard to do a version check and say that you need to upgrade the server? Also it's a minor version, in every software x.x.1 version is a minor change and don't need a complete update of other components. So if upgrading the client breaks the connection with the server you should use a major version number or find a way to maintain compatibility.

kennethjiang, 2022-09-02:
It adds quite a lot of overhead to version both server and 3 clients (OctoPrint plugin, moonraker agent, and the mobile app) and have a huge compatibility matrix. That's why we made the decision to not have the server to be forward compatible. It  cuts out 90% or more of the compatibility complexity. 

Feel free to give it a shot and submit a PR (you probably need multiple PRs since it'll probably need to cover both the server , the octoprint plugin, and the moonraker-obico agent) if you have a good idea on ensuring compatibility.

lettore, 2022-09-02:
It's not a problem if the server is not forward compatible. The problem is that the plugin do a test check and the result is OK but the printer on server is offline. This is really a bad behavior, it should not return OK but say upgrade the server. It's really simple. 
Another problem, where you can find the version of the server? On the web gui is not stated and it should be. Also if I want to build a specific version there's no tag or release on GitHub. How it's supposed to keep the server updated if it's impossible to know which version is running? I really hope that in the future you will use correctly tags and releases on Github, and publish already built Docker images with the correct tag version.

closed 2022-09-02 by lettore

kennethjiang, 2022-09-03:
Like I said we don't version the server. We assume the server is always kept up to date. This assumption is what eliminates the needs for forward compatibility and hence 90% of the complexity.

The test button only tests the connectivity. There are problems that are covered by the diagnostic page but not the test button. There are also plenty of potential problems that are not cover by either of them.

If you think some important cases should be covered in the test button, feel free to take a stab at it and submit a PR to us.