TheSpaghettiDetective/OctoPrint-Obico #186: [BUG] Webcam streaming only updates on reload

3DCoded, 2022-11-01:
**Describe the bug**
When using Obico with OctoPrint, it shows the camera as a snapshot, and only updates when I reload it. On my OctoPrint, it shows the camera as a stream. Why is Obico not streaming the camera?

**To Reproduce**
Steps to reproduce the behavior:
1. Install OctoPrint Docker
2. Install Obico Server
3. Link Both
4. Webcam doesn't update until reload

**Desktop (please complete the following information):**
 - OS: macOS 13.0
 - Browser: firefox, safari
 - Version: latest

Kamoool, 2022-11-02:
What machine are you running your Obico Client on?

kennethjiang, 2022-11-03:
Is this the same as another issue you opened https://github.com/TheSpaghettiDetective/obico-server/issues/674 ? @3DCoded