TheSpaghettiDetective/OctoPrint-Obico #200: [BUG] Push notifications for filament runout no longer working for Obico App

MaliciousCodez05, 2023-03-11:
Push notifications for the Obico app when a printer triggers a filament runout are no longer working for Pusa MK3S+ Printers. 

App Version: v 1.85.157 : b229
Octoprint 1.8.6
Obico v 2.3.1

Emails still work, but not the app notifications.

MaliciousCodez05, 2023-03-11:
Update - getting continual filament runout notifications every minute or so on the following versions: 

Printer: Prusa MK3S+
App Version: v 1.853159 : b231
Octoprint: 1.8.6
Obico: 2.3.0

So it looks like Obico 2.3.1 is where the push notifications on the app broke.

kennethjiang, 2023-03-15:
2.3.0 has a bug that sends repeating notifications. It is fixed in 2.3.1. Maybe the fix caused the push notification to not be sent...?

MaliciousCodez05, 2023-03-21:
> 2.3.0 has a bug that sends repeating notifications. It is fixed in 2.3.1. Maybe the fix caused the push notification to not be sent...?

Yeah, it looks like 2.3.1 fixed the repeating notifications but it broke any notifications for filament runout / filament change.

MaliciousCodez05, 2023-07-08:
> 2.3.0 has a bug that sends repeating notifications. It is fixed in 2.3.1. Maybe the fix caused the push notification to not be sent...?

Any updates @kennethjiang ?

kennethjiang, 2023-07-08:
Sorry we have been tied up with other tasks, primarily the work to rewriting webcam streaming...

Now our new webcam streaming may be finished soon. We will come back to tackling this after that. Sorry for the delay!

MaliciousCodez05, 2023-07-08:
> Sorry we have been tied up with other tasks, primarily the work to rewriting webcam streaming...
> 
> Now our new webcam streaming may be finished soon. We will come back to tackling this after that. Sorry for the delay!

No problem, thank you!

MaliciousCodez05, 2023-10-08:
> Sorry we have been tied up with other tasks, primarily the work to rewriting webcam streaming...
> 
> Now our new webcam streaming may be finished soon. We will come back to tackling this after that. Sorry for the delay!

@kennethjiang Just bumping this to see if there's been any progress on fixing the push notifications for filament runout in the app. Thanks!

kennethjiang, 2023-10-08:
@MaliciousCodez05 Sorry for taking so long. I just got a chance to run a few tests on my Prusa Mini and the notification change came in as expected.

Do you want to come to our discord server so that we can do some debugging together? https://obico.io/discord/

kennethjiang, 2023-10-09:
@MaliciousCodez05 I think I found the problem and fixed it. Can you confirm?

MaliciousCodez05, 2023-10-17:
> @MaliciousCodez05 I think I found the problem and fixed it. Can you confirm?

@kennethjiang Everthing works great now, thank you!

closed 2023-10-17 by MaliciousCodez05