TheSpaghettiDetective/OctoPrint-Obico #203: [Feature] Multistream Capability

cr45hmurphy, 2023-04-15:
I would like to use the YouTube Live plugin for Octopi, but I have to choose between using either Obico or YT Live, due to what I assume is overlap in the usage of the camera.

**Describe the solution you'd like**
Ability to either perform multi-streams without stepping on each other, or ability to push Orbico stream to YT or Twitch.

kennethjiang, 2023-04-18:
Have you tried to set the compatibility mode to "always"? https://obico.io/docs/user-guides/octoprint-webcam-not-streaming/#2-change-settings-to-always-stream-in-compatibility-mode

cr45hmurphy, 2023-04-18:
Yes. Have had that set since I started using Obico.

kennethjiang, 2023-04-18:
In compatibility mode, Obico doesn't take control of the webcam. So I'm not sure what we can further do here. Is there any error?

cr45hmurphy, 2023-04-18:
Obico states that it cannot view or use the webstream when sending it to the YT Live plugin.

kennethjiang, 2023-04-21:
Can you post a screenshot? I can't recall any place in our code that specifically check for YT live plugin.

cr45hmurphy, 2023-06-10:
I apologize for not replying to this sooner. The notification about your reply seemed to get lost in the mix.

Let me see if I can grab a screenshot. I never receive an actual error; it simply won't send to Obico if I am using the YT Live plugin. If I stop that plugin, Obico works again.