TheSpaghettiDetective/OctoPrint-Obico #206: [BUG] Plugin is adding M117 lines to gcode that is uploaded

timur-tabi, 2023-05-19:
I'm not sure how, but it appears that the Obico plugin is adding "M117 DASHBOARD_LAYER_INDICATOR xxx" lines to the gcode as they are being uploaded from PrusaSlicer.  This M117 command is causing a useless message to appear on the printer's control panel (I have a Prusa MK3s).

Here is a file uploaded to Octoprint when the Obico plugin is disabled (I had to rename the file to .txt to attach it):
[TileGrid_Element__20230519090333.txt](https://github.com/TheSpaghettiDetective/OctoPrint-Obico/files/11517676/TileGrid_Element__20230519090333.txt)

And here is the same file when it's uploaded with Obico enabled:
[TileGrid_Element__20230518183617.txt](https://github.com/TheSpaghettiDetective/OctoPrint-Obico/files/11517688/TileGrid_Element__20230518183617.txt)

If you compare the two, you will see that the file uploaded when Obico is enabled as a bunch of lines like this:
M117 DASHBOARD_LAYER_INDICATOR 0
M117 DASHBOARD_LAYER_INDICATOR 1
M117 DASHBOARD_LAYER_INDICATOR 2

This appear on every layer change.  So during the print, the bottom of my printer's console just says "DASHBOARD_LAYER_INDI" during the entire print.

I'm running Octoprint Version 1.8.7 on a Raspberry Pi 3B+.  All plugins are the latest versions.

aarontuftss, 2023-05-22:
Hey! 

This behavior is actually expected.. I apologize if the control panel messages are frustrating. We took inspiration from the popular OctoPrint Dashboard plugin ( [https://github.com/j7126/OctoPrint-Dashboard](url) ) & implemented a layer progress tracking feature almost identical to theirs. 

You can try installing their plugin & see if the same control panel messages are present. You can also view the layer progress in the mobile app - live as of this morning.

Let me know if you have any other questions / concerns.

timur-tabi, 2023-05-22:
So you're saying replacing the bottom line of my control panel screen with useless, truncated text is a feature?

I guess it's time to look at alternatives to Obico.

kennethjiang, 2023-05-22:
The feature is to show the layer during the print, which many users have asked for.

We missed the side effect of this feature in our development. And we will learn from it and do better the next time. We can also discuss if it's a good idea to have an option to enable/disable this feature. However, if you believe what we did was just to mess with our users with useless info, then you don't have enough faith in the Obico team and it's probably a good idea for you to look for an alternative solution

timur-tabi, 2023-05-22:
My point is that this feature doesn't work on a Prusa MK3 because the line

M117 DASHBOARD_LAYER_INDICATOR 0

is much too long.  And why the weird string "DASHBOARD_LAYER_INDICATOR"?  Why not just write "M117 Layer 0"?

I also think it's strange that this happens when the gcode is uploaded to Octoprint, instead of during the print itself.

And finally, why copy features from other plugins that have nothing to do with Obico's main purpose?  I use Obico to monitor my print remotely.  I don't need any features that only work if I'm at the printer itself.

aarontuftss, 2023-05-22:
Sure, the string could be abbreviated to handle the case in which the message is displayed on the printer's LCD to prevent confusion. Although, displaying the message locally is not the feature.

This addition allows you to view which layer your printer is currently working on from either the mobile or web app, no matter where you are at. This directly aligns with Obico's main purpose of remote monitoring - please start a print & view progress in mobile app for reference.

As mentioned above, this is something many users have asked for & we are accommodating.

timur-tabi, 2023-05-22:
The M117 command is only supposed to change the LCD console on the printer itself.  That's literally how it's defined.  If you want to display progress on some app, you need to use a different method.

This "feature" should be an option, not something that's forced on users.

matusmisiak, 2023-05-23:
I agree it is very useful feature and I really appreciate your work. However, as @timur-tabi wrote too, the use of M117 for this case is quite confusing for me.

1. What if I wanted to display my own or other's plugin messages? It could be great if I could turn off the addition of M117.
2. Even printers that don't truncate the message (e.g. Ender 3) usually scroll the text, so it takes at least 10 seconds to see the actual layer number.
3. Maybe I just missed it, but I haven't seen this feature in the changelog. I spent quite a while finding out why has the status line changed :)

cp2004, 2023-05-23:
> We took inspiration from the popular OctoPrint Dashboard plugin ( https://github.com/j7126/OctoPrint-Dashboard ) & implemented a layer progress tracking feature almost identical to theirs.

The dashboard plugin does prevent these lines from ever actually being sent to the printer - they are only used for tracking the layers internally to the plugin, and so the users don't have a problem.

aarontuftss, 2023-05-23:
Oh interesting.. after taking another look i see now in their code where they remove the M117 so the message does not actually get sent to the printer for display.

I really appreciate your guys's feedback. We will be making some changes to further improve this experience !

aarontuftss, 2023-05-24:
Hey all, just pushed up a fix to our dev branch. If any of you guys could validate the solution, that would be very helpful. 

My setup does not respond to the M117 command so I can only do a limited amount of testing locally.

[https://github.com/TheSpaghettiDetective/OctoPrint-Obico/tree/dev](url)

kennethjiang, 2023-05-24:
> Hey all, just pushed up a fix to our dev branch. If any of you guys could validate the solution, that would be very helpful.
> 
> My setup does not respond to the M117 command so I can only do a limited amount of testing locally.
> 
> https://github.com/TheSpaghettiDetective/OctoPrint-Obico/tree/dev

To clarify what @aarontuftss means here: please go to OctoPrint settings dialog -> plugin manager -> "+Get More" -> "from URL", and use this URL: `https://github.com/TheSpaghettiDetective/OctoPrint-Obico/archive/refs/heads/dev.zip` .

@timur-tabi @matusmisiak it'll be great if you can help us verify if the fix works. Somehow between about a dozen printers between all Obico team members, all of them ignored M117 gcode. That's why we will need your help to verify the fix. Thanks!

matusmisiak, 2023-05-24:
Yes, it works for me. Thank you for your quick response!

kennethjiang, 2023-05-24:
Closed by https://github.com/TheSpaghettiDetective/OctoPrint-Obico/commit/a723453acdd66e3a3a39d03af0b5a3d3416ae77c

matusmisiak, 2023-05-24:
I am courious about that too. I understand this functionality was taken from the dashboard plugin, but why can't it parse directly the layer change comment while printing?

aarontuftss, 2023-05-24:
Layer changes are marked with non-gcode commands / comments on their own line within the file like `;LAYER:([0-9]+)` & a few other variations but its dependent on the slicer. 

These lines are not picked up by any of OctoPrints available [hooks](https://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-phase) that are fired during the print process. So we have to insert our own indicator upon file upload using OctoPrint's [preprocessor](https://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-filemanager-preprocessor) hook that dissects the file line by line where we can find these indicators & insert our own that will be picked up later on.

Adding the option to toggle this is something we can add to our roadmap & prioritize accordingly. Also, I can spend some reviewing other possible solutions.

matusmisiak, 2023-05-24:
Thanks, makes sense!

timur-tabi, 2023-05-25:
I'm a bit concerned that you overload M117 for this purpose.  I know that's what Dashboard does, but it still feels wrong.  What if I have some other plugin or slicer setting that uses M117 for legitimate purposes?

I suggest that you pick a difference mcode, one that isn't used by any printer.  Perhaps even add an option for users to specify.

However, I'm guessing what's really needed is for Octoprint to provide a layer change callback to plugins.

aarontuftss, 2023-05-25:
I understand.. We do not prevent _every_ M117 from being passed to printer, only the specific lines we injected using a regular expression check. 

Upon this thread being opened, it had actually crossed my mind to use a dummy mcode that does nothing (like M2000). Something thats not a real command but will still get picked up during the hooks. I can play with the idea.

cp2004, 2023-05-25:
> However, I'm guessing what's really needed is for Octoprint to provide a layer change callback to plugins.

OctoPrint doesn't know when the layer changes (it does already have events for notifying when the Z height changes), and since the implementations in plugins rely on parsing comments out of the gcode that there is no set standard for we won't add it to OctoPrint's core.

timur-tabi, 2023-05-26:
May I suggest that you file a feature request with Octoprint to add this functionality?  It could do something similar to what the Dashboard plugin does: scan the gcode for patterns based on the slicer.

wademaxfield, 2023-06-07:
This addition is messing with the Layer Progress gauge displayed on Dashboard.  Some other process (Dashboard?) is also writing a M117 DASHBOARD_LAYER_INDICATOR line. 

Downloading an uploaded .gcode shows lines like:

```
;LAYER:9
M117 DASHBOARD_LAYER_INDICATOR 9
M117 DASHBOARD_LAYER_INDICATOR 10
 ```

Disabling Obico, uploading a freshly sliced .gcode, then downloading that from Octoprint shows only a single M117 DASHBOARD_LAYER_INDICATOR
```
;LAYER:9
M117 DASHBOARD_LAYER_INDICATOR 9
```

So my assumption is that Obico is adding the 2nd M117 DASHBOARD_LAYER_INDICATOR.  In addition, the line that Obico was adding was incorrectly numbered.  I am using cura which starts its numbering at 0.  I am guessing that Obico was not looking at the layer number from th comment, but using its own count of ;LAYER comments instead.

These 2 M117 DASHBOARD_LAYER_INDICATOR lines are causing the Layer % gauge on dashboard to calculate incorrectly, printing a 20mm calibration cube at 0.15mm, and my gauge is up to 203000% by the time it reaches layer 103.  It used to go to 100% then reset to 0% when the next layer started.

Please add an option to disable this new functionality for those of us already using a plugin that also does this.

aarontuftss, 2023-06-08:
I see. Will have a fix up here soon, thank you

timur-tabi, 2023-06-25:
Why is this bug report closed if the bug hasn't been fixed yet?  My printer console still says DASHBOARD_LAYER_INDI even after the print is finished!

aarontuftss, 2023-06-26:
Please update your plugin to 2.4.1

If you are reprinting a file that was uploaded on an older version, please re-upload the file.

timur-tabi, 2023-06-26:
Ugh, this is why I don't like the fact that it modifies the gcode upon uploading.  Old bugs corrupt my gcode forcing me to reslice the file.

I think I'm just going to delete the plugin.  I can't remember the last time it actually detected a real print problem and notified me on time anyway.

lightmaster, 2024-01-08:
I'm on plugin version 2.4.10 and I'm still seeing M117 messages scroll across my printer's screen while printing. On many prints it takes longer to finish scrolling than the layer actually takes to print, so my screen is just constantly filled with "M117 OBICO_LAYER_INDICATOR 1" over and over again. Yes, I do have the referenced Dashboard plugin installed and I do not see any messages from it, only Obico's message and the occasional message from DisplayLayerProgress when the layer is actually long enough for Obico's message to get done scrolling.

I still see no method of disabling this setting that taints any gcode that was uploaded while the plugin was enabled and have to have the plugin disabled, reslice, and re-upload any gcode I want to print in order to correct the corrupted gcode. Seeing as how it massively messes with any screen connected to the printer, yes "corrupted" is the correct term.