TheSpaghettiDetective/OctoPrint-Obico #208: [BUG] Brief bug description here

reuter, 2023-05-26:
After upgrading to 1.9.0 (I'm running on Ubuntu, not a Pi) Obico is now erroring out when starting Octoprint. The cameras are now loading.

**To Reproduce**
Steps to reproduce the behavior:
Boot Octoprint
Error pResents itself
```

2023-05-26 02:04:31,023 - backoff - INFO - Backing off capture_jpeg(...) for 0.9s (urllib.error.URLError: <urlopen error [Errno 111] Connection refused>)
2023-05-26 02:04:31,974 - backoff - INFO - Backing off capture_jpeg(...) for 0.3s (urllib.error.URLError: <urlopen error [Errno 111] Connection refused>)
2023-05-26 02:04:32,262 - backoff - ERROR - Giving up capture_jpeg(...) after 3 tries (urllib.error.URLError: <urlopen error [Errno 111] Connection refused>)
2023-05-26 02:04:32,263 - octoprint.plugins.obico - WARNING - Failed to capture jpeg - <urlopen error [Errno 111] Connection refused>
2023-05-26 02:04:41,275 - backoff - INFO - Backing off capture_jpeg(...) for 0.6s (urllib.error.URLError: <urlopen error [Errno 111] Connection refused>)
2023-05-26 02:04:41,834 - backoff - INFO - Backing off capture_jpeg(...) for 0.4s (urllib.error.URLError: <urlopen error [Errno 111] Connection refused>)
2023-05-26 02:04:42,205 - backoff - ERROR - Giving up capture_jpeg(...) after 3 tries (urllib.error.URLError: <urlopen error [Errno 111] Connection refused>)
2023-05-26 02:04:42,206 - octoprint.plugins.obico - WARNING - Failed to capture jpeg - <urlopen error [Errno 111] Connection refused>
2023-05-26 02:04:52,217 - backoff - INFO - Backing off capture_jpeg(...) for 0.8s (urllib.error.URLError: <urlopen error [Errno 111] Connection refused>)
2023-05-26 02:04:53,065 - backoff - INFO - Backing off capture_jpeg(...) for 1.0s (urllib.error.URLError: <urlopen error [Errno 111] Connection refused>)
2023-05-26 02:04:54,078 - backoff - ERROR - Giving up capture_jpeg(...) after 3 tries (urllib.error.URLError: <urlopen error [Errno 111] Connection refused>)
2023-05-26 02:04:54,079 - octoprint.plugins.obico - WARNING - Failed to capture jpeg - <urlopen error [Errno 111] Connection refused>
2023-05-26 02:05:00,533 - octoprint.events.fire - DEBUG - Firing event: DisplayLayerProgress_timerTrigger (Payload: {'updateReason': 'timerTrigger', 'totalLayer': '21', 'currentLayer': '-', 'lastLayerDuration': '-', 'lastLayerDurationInSeconds': '-', 'averageLayerDuration': '-', 'averageLayerDurationInSeconds': '-', 'currentHeight': '0.40', 'currentHeightFormatted': '0.4', 'totalHeight': '7.2', 'totalHeightFormatted': '7.2', 'feedrate': '1000', 'feedrateG0': '-', 'feedrateG1': '1000', 'fanspeed': '-', 'progress': '0', 'm73progress': '0', 'printTimeLeft': '1h46m', 'printTimeLeftInSeconds': 6360, 'printerState': 'printing', 'estimatedEndTime': '03:51', 'estimatedChangedFilamentTime': '-', 'changeFilamentTimeLeft': '-', 'changeFilamentTimeLeftInSeconds': 0, 'changeFilamentCount': 0, 'currentFilename': 'House_0.6n_0.35mm_PETG_MK3SMMU2S_1h47m.gcode'})

```

**Screenshots**
[image]

kennethjiang, 2023-05-29:
This should be fixed by version 2.3.9. Feel free to re-open it if the problem persists after you upgrade to 2.3.9.

closed 2023-05-29 by kennethjiang

reuter, 2023-05-29:
That fixed it, thanks.
Sorry about the title... got ahead of myself filling out the bug report :D

kennethjiang, 2023-05-29:
Haha no problem at all @reuter