TheSpaghettiDetective/OctoPrint-Obico #217: [BUG] Possible memory leak with Obico `ffmpeg` video streaming

calexandre, 2023-11-25:
**Describe the bug**

I've been stugling with an instability issue on my OctoPrint instance, where from time to time the system would run out of memory.
After some debugging, I've managed to isolate the issue to the Obico plugin, specifically the `ffmpeg` process.

- It seems that when I have the streaming enabled (which is the purpose of Obico), the `ffmpeg` process is automatically spawned and will slowly consume memory until OctoPrint ends up crashing due to out of memory.
- This will take some days until memory pressure starts causing instability in the entire system, which will eventually cause OctoPrint to crash and fail the print.

I've managed to isolate this issue by letting the system running some time **with and without** the Obico plugin enabled, which led me to open this issue.

To mitigate this issue, i've setup a cron to kill the `ffmpeg` process every day.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable Obico streaming
2. Monitor the `ffmpeg` process and check the memory it is consuming
3. After a couple of days, you go back to the `ffmpeg` process, and notice the sudden increase in memory, and depending on the interval, you may be already having swaping problems due to lack of available memory
4. If you dont kill the `ffmpeg` process, OctoPrint will eventually crash

**Screenshots**

Day 0 - Rebooted everything, notice the `ffmpeg` memory %

[image]

Day 1 - After 27 hours, the `ffmpeg` process is already taking up almost 30% of memory

[image]

**Desktop (please complete the following information):**
 - OS: [OctoPi 1.0.0 & OctoPrint 1.9.3 (new camera stack)](https://github.com/OctoPrint/OctoPi-UpToDate/releases/download/1.0.0-1.9.3-20231009154319/octopi-1.0.0-1.9.3-20231009154319.zip)
 - Browser Chrome
 - Version Obico for OctoPrint (2.4.9)

**Additional context**
- I'm a paying customer of the Obico solution, so I have premium streaming enabled.
- I don't know if it is related, but I reported another issue related to streaming issues: https://github.com/TheSpaghettiDetective/OctoPrint-Obico/issues/216

kennethjiang, 2023-12-01:
Sorry I have been really busy post-Thanksgiving.

This sounds like a pretty bad problem. However I haven't been able to reproduce it.

I notice you are using RTSP stream provided by OctoPrint's new webcam streaming stack. Can you disable it (do the opposite of https://community.octoprint.org/t/got-rtsp-on-octopi-working/54091), restart the Pi, and see if ffmpeg will still have memory leak?

Also do you remember if this problem has always been there, or it started happening at some point? If the latter, do you remember if there was any change to your setup right before this started to happen?

kennethjiang, 2023-12-09:
It sounds like the issue has been going on for a while now.

I guess you were using the octopi 1.0 SD card image? If so, have you done system upgrade? I'm wondering if it's possible that a system lib was upgraded to a version that causes the memory leak.