TheSpaghettiDetective/OctoPrint-Obico #220: [Feature] Allow Obico to Reprint from GCode stored on Octoprint-server if not available on Obico-server

puterboy, 2023-12-24:
Currently, it seems that Obico can only "reprint" if you originally had uploaded the GCode to Obco-server but not if you had launched the print from Octoprint.

It would be great if the 'Reprint' feature would work more generally so that:

- If the original print was uploaded to Obico, then the reprint would use the GCode stored on the Obico-server
- If the original print was launche from Octoprint, then the reprint would use the GCode store on the Octoprint-server

This should not be too hard to do since Obco is already using the Octoprint api to allow users to *browse* Gcode files stored on Octoprint.
The implementation logic would work roughly as follows.

1. Use the `resident_printer_id` stored in the `app_gcodefile` table to determine if the GCODE was uploaded to Obico or started with Octoprint (and then this is the Printer ID) [From my chat with @Apex, I believe this is what the field does]
2. If the GCode was uploaded to Obico, then continue with current behavior
3. If the GCode was printed directly on Octoprint, then use the `url` field in the `app_gcodefile` table to launch the print from Octoprint (using the api similarly to how it is done now when you are in the "Printers" tab and press "Octoprint" to "Open G-Code File to Start Printing").  If GCode file no longer present return a "file not found" error.
4. Optionally, check the current stored GCode file size and md5sum for consistency to make sure GCode file is the same.

There are multiple reasons why one may want to load your initial file from Octoprint but later want to reprint from Obico.
For example, most of the time I launch my printing from Prusa Slicer which has an automatic upload to Octoprint (and I don't think it can upload directly to Obico).

At the same time, I use Obico to monitor prints and I prefer it's print history to Octoprints -- so I often use Obico to identify a historical print that I want to reprint.

kennethjiang, 2023-12-27:
We don't have the re-print button for gcode files in octoprint or klipper because there will be too many edge cases, such as:

- The gcode file is a/b/c.gcode in octoprint/klipper, but it was later deleted.
- The gcode file is a/b/c.gcode in octoprint/klipper, but it was later overwritten with a different version. The "reprint" button will print a different version, which may or may not be a surprise for the user.
- The gcode file is a/b/c.gcode in octoprint/klipper, but it was later moved to x/y/z.gcode. No matter what we do here, the result will be a surprise for the users.
- The gcode file is a/b/c.gcode in octoprint/klipper, but later overwritten with a different gcode file. "Reprint" will result in a wrong behavior.
- ...

Before we have a good way to handle edge cases like these, we can't add the "reprint" function.

puterboy, 2023-12-27:
At least for Octoprint, all of the above edge cases could be mitigated if not solved if Obico stored the md5sum of the gcode in the `Print History` which could then be checked against the md5sum hash stored in the corresponding `.metadata.json` file.
This hash is accessible via the Octoprint API.

Basically, Obico would query the api for the file based on its last known location and check the hash. If they agree, then the reprint  button is enabled. Otherwise it's not.

puterboy, 2023-12-27:
Alternatively, while not as robust, you could use the existing `num_bytes` field from the `app_gcodefile` table and the last print time from the `app_print` table to match against similar values availabe from the Octoprint API.

Of course, I understand that this might not be a priority at all - but technically I don't see any big conceptual issues to getting this right almost always.

And even if not perfect, it would still be better than the Octoprint list of gcodes that is much harder to navigate and silently overwrites itself so long as the name/print-time/filament-type are the same.

kennethjiang, 2023-12-30:
Klipper is trickier, but it is still technically possible. Still, the user experience will be bad - in case of md5 mismatch, how do we explain it to the user?