TheSpaghettiDetective/OctoPrint-Obico #225: [BUG] Automatically downgrades to ffmpeg_from_mjpeg if you don't have webcamd installed or if you don't have sudo access to stop service

puterboy, 2024-01-21:
The current code in `webcam_stream.py` assumes that:

1. The user has `webcamd` installed as a service
2. The use has 'sudo' privileges to shutdown the service

Otherwise, the following 'if' statement will cause the streaming to revert to 'ffmpeg_from_mjpeg' even if you have a valid stream already running.
```
if sarge.run('sudo service webcamd stop').returncodes[0] != 0:
                self.ffmpeg_from_mjpeg()
                return
```

I for example use Janus to generate my webcam stream directly from the native RPi camera mp4 stream. Others may want to use their own streaming program.
Also, I successfully run both Obico and Octoprint without sudo permissions.

The patch for this is simple:
```
                 self.ffmpeg_from_mjpeg()
                 return

-            if sarge.run('sudo service webcamd stop').returncodes[0] != 0:
+            if sarge.run('systemctl is-active webcamd').returncodes[0] == 0 and sarge.run('sudo service webcamd stop').returncodes[0] != 0:
                 self.ffmpeg_from_mjpeg()
                 return
```

Note that there are other places in `webcam_stream.py' where the use of `webcamd` is assumed if the service needs to be started. These should probably be addressed to for other use cases. But it doesn't affect the use case where you have a valid stream already.

I created a simple pull request.

kennethjiang, 2024-01-22:
> The current code in `webcam_stream.py` assumes that:
> 
>     1. The user has `webcamd` installed as a service
> 
>     2. The use has 'sudo' privileges to shutdown the service

Since the code is `if sarge.run('sudo service webcamd stop').returncodes[0] != 0:`, it doesn't assume any of these. If falls back to `ffmpeg_from_mjpeg` if any of these conditions are not met.

puterboy, 2024-01-22:
I opened one already :) Let me know if you don't see it.
The diagrams I shared I think should be helpful in understanding the flow.

BTW, my PR just adds an additional to test to the line of code to check first if `wbcamd` is active because if it's inactive no point in trying to try to shut it down (and then fail) -- indeed, quite the opposite - the fact that `webcamd` is not active probably means that you have a good native h264 stream.

puterboy, 2024-06-19:
This is fixed in new `webcam_stream.py` that no longer uses `sarge` or messes with `webcamd` service.

closed 2024-06-19 by puterboy