TheSpaghettiDetective/OctoPrint-Obico #231: Multiple account logon capability in app

nigeljh99, 2024-03-15:
Having two premium accounts, one for home and one for work, it would be very useful if the app allowed me to logon to both and stay logged in to both.  I could then switch between the different sets of printers being displayed .
Perhaps even allow all printers form multiple accounts to be displayed at once, but that may be a step too far for security concerns and development at the moment.

kennethjiang, 2024-03-21:
Sorry for the delay! Was traveling until last week.

The feature request here makes a lot of sense. This feature make take a while to implement, as it will touch almost every part in the app. For now, you can log in one account in the Obico mobile app, but log in another account in a mobile browser. Do you think this workaround is feasible?

nigeljh99, 2024-04-11:
Thanks Kenneth, yes, that's what i have been doing really, and thank you for your comment. Appreciate it could be a big task!