TheSpaghettiDetective/OctoPrint-Obico #239: [BUG] Webcam Streaming Failed notification even when Webcam Streaming disabled

TarheelGrad1998, 2025-02-12:
I have moved to running OctoPrint in a docker container, and was unable to get the Webcam Streaming within my self hosted Obico to work due to known limitations, per my understanding.  This is fine, I have plenty of ways to access the streams, and it does not appear to affect detection.  So, I disabled Webcam Streaming through the plugin settings.

Despite this, every time I log into the Obico webpage, I still get the "Obico for OctoPrint: Webcam Streaming Failed" notification.  Why?  This seems like a bug to me.

kennethjiang, 2025-02-16:
Sorry for the delay. It does sound like a bug. Can you describe the steps to reproduce this bug?

TarheelGrad1998, 2025-02-17:
It's not much more than what's in the OP:

1. Set up OctoPrint in Docker
2. Set up Obico in Docker
3. Install Obico Plugin in OctoPrint and link to Obico
4. Login to Obico -- Notification received
Obico for OctoPrint: Webcam Streaming Failed
Follow the webcam troubleshooting guide to resolve the issue.
5. Access Obico Plugin settings, check disable webcam streaming
6. Login to Obico -- still receive above Notification
At this point, every login will trigger the above Notification.

Potentially relevant log entries:

2025-02-07 22:46:51,531 - octoprint.plugins.obico - WARNING - No ffmpeg found, or ffmpeg does NOT support h264_omx/h264_v4l2m2m encoding.
2025-02-07 22:46:51,547 - octoprint.plugins.obico - ERROR - Janus not found or not configured correctly. Quiting webcam streaming.
2025-02-07 22:46:51,549 - octoprint.plugins.obico - DEBUG - Sending to server: 
{'passthru': {'printer_event': {'event_title': 'Obico for OctoPrint: Webcam Streaming Failed', 'event_text': 'Follow the webcam troubleshooting guide to resolve the issue.', 'event_class': 'WARNING', 'event_type': 'PRINTER_ERROR', 'info_url': 'https://obico.io/docs/user-guides/webcam-feed-is-not-showing/'}}}
2025-02-07 22:46:51,559 - octoprint.plugins.obico - DEBUG - API called: get_plugin_status

kennethjiang, 2025-02-18:
@TarheelGrad1998 thank you for the details. I pushed the fix to the "dev" branch. Can you help me test if the fix works for you? 

you can test the "dev" branch by:

OctoPrint settings -> Plugin manager -> Get More -> From URL ->

https://github.com/TheSpaghettiDetective/OctoPrint-Obico/archive/refs/heads/dev.zip

closed 2025-02-18 by kennethjiang